united-manufacturing-hub/united-manufacturing-hub #1068: Bug Report: valid addProduct messages land up in putback

JeremyTheocharis, 2022-05-06:
### Expected Behavior

addProduct messages should land up in database

### Current Behavior

Specific messages landed up in putback

```json
{
  "product_id":"product_nhycw30pft"
  "time_per_unit_in_seconds":1
}
```

```json
{
  "Putback":{
    "FirstTsMs":1651847341588
    "LastTsMs":1651847642539
    "Amount":314
    "Reason":"Other"
  }
}
```

### Steps to Reproduce

?

### Context (Environment)

Venice customer

### Version

v0.9.x

Scarjit, 2022-05-06:
Tracking logging for this in #1069

Scarjit, 2022-05-13:
@JeremyTheocharis This was released in 0.9.3, has the release been deployed at Venice, and was there any observed misbehaving traffic afterwards ?

Scarjit, 2022-06-03:
@JeremyTheocharis Where you able to observe this behaviour, after 0.9.3 ?

closed 2023-03-31 by Scarjit