united-manufacturing-hub/united-manufacturing-hub #1172: Inconsistency: The message prefixes Activity and DetectedAnomaly are not in kafka-to-postgresql

Sphingobium, 2022-06-14:
### Expected Behavior

The message prefixes "Activity" and "detectedAnomaly" should be present in the kafka-to-postgresql  prefix.go file and also have their own px<>.go file 

### Current Behavior

The message prefixes activity and detectedAnomaly are documented on docs and soon to be documented on learn, but do not show up in the prefix.go of kafka-to-postgresql and also dont have a px.go file associated to them.

### Steps to Reproduce

look into the documentation and see activity as documented message topic.

look into kafka-to-postgresql and look for activity and don't find it.

### Context (Environment)

I am currenty writing the documentation on learn.umh.app and stumbled on this issues as i was documenting.

### Version

v0.9.x

Scarjit, 2022-06-14:
@JeremyTheocharis These two where neither in mqtt-to-postgresql nor in kafka-to-postgresql, are they actually relevant ?

JeremyTheocharis, 2022-06-15:
Can be closed. Will be used in the future by kafka-state-detector

closed 2022-06-15 by JeremyTheocharis