united-manufacturing-hub/united-manufacturing-hub #1173: Timescale can result in server failure due to high disk usage

Scarjit, 2022-06-14:
### Expected Behavior

Timescale should automatically cleanup data like processValue after X days.

This can be done using timescales dataretention feature:
https://docs.timescale.com/timescaledb/latest/how-to-guides/data-retention/#data-retention=

### Current Behavior

Timescale does not cleanup old data, resulting in full disks.

### Steps to Reproduce

1) Setup stack
2) Add data ingestion
3) Wait

### Context (Environment)

Occurred on dev-server

### Possible Solution

https://docs.timescale.com/timescaledb/latest/how-to-guides/data-retention/#data-retention=

### Version

v0.9.x

Scarjit, 2022-06-14:
@JeremyTheocharis We should discuss a good retention period for processValue

Scarjit, 2022-06-14:
[image]
Before DELETE + VACUUM -> After

JeremyTheocharis, 2022-06-15:
By default, we should not have a retention period as this might be unexpected for some customers.

Scarjit, 2022-06-24:
https://github.com/united-manufacturing-hub/learn.umh.app/issues/21

closed 2022-06-30 by JeremyTheocharis