united-manufacturing-hub/united-manufacturing-hub #1176: Bug Report - Auto Order Add Problem

onurcankaraman, 2022-06-16:
### Expected Behavior

Adding the order added with AddOrder to the database only once.

### Current Behavior

Hello, when I add a product with add order once, the order is added again at certain intervals. I'm pretty sure there is no case of repeatedly running code over Node-Red.

### Steps to Reproduce

Step 1. Products were created on Node-Red.
Step 2. Order added using inject.
Step 3. The database was checked and it was seen that the order was added again with certain steps.

### Context (Environment)

When I want to show completed orders on grafana, it doesn't show up correctly.

### Possible Solution

My guess is that something in the background is listing orders and keeps adding them over and over.

### Version

v0.9.x

### Anything else?

[united-manufacturing-hub-mqtttopostgresql-0.log](https://github.com/united-manufacturing-hub/united-manufacturing-hub/files/8918502/united-manufacturing-hub-mqtttopostgresql-0.log)
[united-manufacturing-hub-nodered-0.log](https://github.com/united-manufacturing-hub/united-manufacturing-hub/files/8918503/united-manufacturing-hub-nodered-0.log)
[united-manufacturing-hub-timescaledb-0.log](https://github.com/united-manufacturing-hub/united-manufacturing-hub/files/8918504/united-manufacturing-hub-timescaledb-0.log)

My Log Files

Scarjit, 2022-06-21:
Hi,
thanks for the report.
It seems that you use mqtt-to-postgresql, which is outdated for a while now (since ~[v0.8.0](https://github.com/united-manufacturing-hub/united-manufacturing-hub/releases/tag/v0.8.0))

Is it possible for you to send us an example of the re-inserted orders inside the db ?

onurcankaraman, 2022-06-22:
[image]

You can see it in the picture.

I removed the relevant block on Node-Red. I can't do add Order right now.

image shows as 0.8.7. UMH Release is 0.9.4 Prerelease 7.

Scarjit, 2022-06-24:
Hi,

is it possible for you to send me the values you are using in k3os ?

onurcankaraman, 2022-06-24:
_000_commonConfig:
  blobstorage:
    enabled: false
  certificates:
    CACert: |-
      -----BEGIN CERTIFICATE-----
      
      -----END CERTIFICATE-----
    barcodereader:
      sslCertificatePem: |-
        -----BEGIN CERTIFICATE-----
       
        -----END CERTIFICATE-----
      sslKeyPassword: ""
      sslKeyPem: |-
        -----BEGIN PRIVATE KEY-----
        
        -----END PRIVATE KEY-----
    kafka:
      cert: |-
        -----BEGIN CERTIFICATE-----

      privkey: |-
        -----BEGIN PRIVATE KEY-----
       
        -----END PRIVATE KEY-----
    kafkabridge:
      sslCACertRemote: |-
        -----BEGIN CERTIFICATE-----
        
        -----END CERTIFICATE-----
      sslCertificatePemLocal: |-
        -----BEGIN CERTIFICATE-----
        -----END CERTIFICATE-----
      sslCertificatePemRemote: |-
        -----BEGIN PRIVATE KEY-----
      
        -----END PRIVATE KEY-----
      sslKeyPasswordLocal: ""
      sslKeyPasswordRemote: ""
      sslKeyPemLocal: |-
        -----BEGIN PRIVATE KEY-----
      
        -----END PRIVATE KEY-----
      sslKeyPemRemote: |-
        -----BEGIN CERTIFICATE-----
       
        -----END CERTIFICATE-----
    kafkadebug:
      sslCertificatePem: |-
        -----BEGIN CERTIFICATE-----
    
        -----END CERTIFICATE-----
      sslKeyPassword: ""
      sslKeyPem: |-
        -----BEGIN PRIVATE KEY-----
        
        -----END PRIVATE KEY-----
    kafkainit:
      sslCertificatePem: |-
        -----BEGIN CERTIFICATE-----
       

        -----END CERTIFICATE-----
      sslKeyPassword: ""
      sslKeyPem: |-
        -----BEGIN PRIVATE KEY-----
        
        -----END PRIVATE KEY-----
    kafkastatedetector:
[... 2672 lines cut ...]
    create: false
    labels: {}
  statefulset:
    annotations: {}
    labels: {}
    lifecycle: {}
    livenessProbe:
      failureThreshold: 3
      initialDelaySeconds: 600
      periodSeconds: 10
      successThreshold: 1
      timeoutSeconds: 5
    podAnnotations: {}
    podManagementPolicy: OrderedReady
    prometheus.io/port: "8888"
    prometheus.io/scrape: "true"
    readinessProbe:
      failureThreshold: 3
      initialDelaySeconds: 90
      periodSeconds: 10
      successThreshold: 1
      timeoutSeconds: 5
    terminationGracePeriodSeconds: 60
    updateStrategy: RollingUpdate
  tolerations: []

JeremyTheocharis, 2022-06-30:
Here the final report:

When sending addOrder messages, the order was duplicated. The United Manufacturing Hub was probably upgraded from an earlier version. As there is no documentation on how to do it, something must have gotten wrong here. This caused mqtt-to-postgresql to be still existing in parallel with kafka-to-postgresql and there was either a bug in mqtt-to-postgresql or the both systems running together were causing an endless loop. After deleting mqtt-to-postgresql statefulset, the problem is fixed.

closed 2022-06-30 by JeremyTheocharis