united-manufacturing-hub/united-manufacturing-hub #1223: Bug Report

JeremyTheocharis, 2022-08-18:
### Expected Behavior

When sending new processValues, I want them to appear immediately in Grafana

### Current Behavior

It takes a couple of minutes until they appear in Grafana

### Steps to Reproduce

1. Create a new processValue and send a message
2. It will take a couple of minutes until it gets selectable in Grafana

### Context (Environment)

-

### Possible Solution

This is likely happening beause the topic creation process in Kafka takes too much time. Maybe we can speed that step up? If not possible, we should add a general warning in the documentation stating that the first message to any MQTT topic might take a couple of minutes to go initially though (but all subsequent messages will be sub-second). @Sphing

### Version

v0.9.x

Scarjit, 2022-08-22:
This should be fixable by lowering [metadata.max.age.ms](https://docs.confluent.io/platform/current/installation/configuration/consumer-configs.html#consumerconfigs_metadata.max.age.ms)

Scarjit, 2022-08-24:
@JeremyTheocharis Can you give me a good time, for metadata.max.age.ms ?
I don't want to go to low.

Azure uses [3 minutes](https://docs.microsoft.com/en-us/azure/event-hubs/apache-kafka-configurations), Confluent uses 5 minutes by default.

Going to low, could block other kafka operations and will increase load on the broker.

Scarjit, 2022-08-29:
Let's use 3m

PR #1249 (2022-09-26): feat: reduce metadata refresh rate to 3 minutes
# Description

Reduced the refresh rate for Kafka's metadata from 5 minutes to 3 in order to detect faster new processValues

Fixes #1223 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# How Has This Been Tested?

All the pods correctly run on our test server and the new topics arrive.

# Checklist:

- [ ] I have made corresponding changes to the documentation
- [x] I have made corresponding changes to the changelog (if needed).
- [ ] I have made corresponding notices to the upgrade instructions (if needed) [... cut]
closed 2022-09-19 by Scarjit