united-manufacturing-hub/united-manufacturing-hub #1226: factoryinsight has long running distinct queries

JeremyTheocharis, 2022-08-22:
### Expected Behavior

Distinct queries should run quick, especially the ones used in factoryinsight to show potential values.

### Current Behavior

They take up to 55 seconds (and probably even more)

### Steps to Reproduce

1. Have a lot of times-series data
2. Try to select on the the tags in Grafana

### Context (Environment)

Beverage customer

### Possible Solution

See also unpublished knowledgebase article: https://www.notion.so/united-manufacturing-hub/ab58fbca87834bfeb7f654af1dbb43fd?v=61b50b6953fd415bb2ac4dc5dccfc109&p=9a3e71ba952a4727b22e23dcabdbbd45&pm=s

### Version

v0.9.x

Scarjit, 2022-08-24:
Can't reproduce.
I have created a new DB with 250 value names (and ~3 million entries), both cte and the normal query run in < 100 ms.

Scarjit, 2022-08-24:
Creating a greater dataset now, to test further

Scarjit, 2022-08-24:
2nd test with ~19mil entries split in 1k value names.
Both queries still completed in < 100ms.

Will do a last run, with a large amount of value names

Scarjit, 2022-08-24:
Last test for today:
27.471.506 entries, with 878053 different value names.

This time the issue occurred.
(~20 sec with both methods)

[image]

After creating a new index (```CREATE INDEX ON processvaluetable(valuename, asset_id) WITH (timescaledb.transaction_per_chunk);```) on the data, it was brought down to ~3 seconds

PR #1261 (2022-10-04): Added index for faster GF queries
# Description

This pr partly fixes #1226 by inserting an index: `CREATE INDEX ON processvaluetable(valuename, asset_id) WITH (timescaledb.transaction_per_chunk);`.

It also introduces gzip compression for any factoryinsight response, decreasing transmitted data from ~30-70% depending on the request.

Also reverts #1193

## Type of change

- [X] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

- [X] Builds
- [X] Responds correctly to Grafana requests

# Checklist:

- [ ] I have made corresponding changes to the documentation
- [ [... cut]
closed 2022-10-04 by Scarjit