united-manufacturing-hub/united-manufacturing-hub #1243: Creating an alert in Grafana results in errors

Scarjit, 2022-09-08:
### Expected Behavior

Grafana should allow alerts without any errors or warnings

### Current Behavior

Grafana throws an "API error" and says "plugin unavailable"

[image]

### Relevant log output

```shell
t=2022-09-08T08:07:38+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:07:38+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=7 size=23 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:08:05+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:08:05+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=2 size=23 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:08:06+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:08:06+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=2 size=23 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:08:28+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:08:28+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=3 size=23 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:09:10+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=GET path=/api/ruler/grafana/api/v1/rules/test status=202 remote_addr=10.42.0.1 time_ms=3 size=2 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:09:10+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/ruler/grafana/api/v1/rules/test status=202 remote_addr=10.42.0.1 time_ms=18 size=45 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:09:10+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=GET path=/api/ruler/grafana/api/v1/rules/test/testa status=202 remote_addr=10.42.0.1 time_ms=2 size=1109 referer=http://10.0.2.1:8080/alerting/new
t=2022-09-08T08:09:13+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:09:13+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=1 size=23 referer=http://10.0.2.1:8080/alerting/PT9b_AM4k/edit
t=2022-09-08T08:10:08+0000 lvl=info msg="level=debug component=dispatcher msg=\"Received alert\" alert=testa[008ac8d][active]" logger=alertmanager org=1
t=2022-09-08T08:10:38+0000 lvl=info msg="level=debug component=dispatcher aggrGroup={}:{} msg=flushing alerts=[testa[008ac8d][active]]" logger=alertmanager org=1
t=2022-09-08T08:10:38+0000 lvl=info msg="level=error component=dispatcher msg=\"Notify for alerts failed\" num_alerts=1 err=\"grafana-default-email/email[0]: notify retry canceled due to unrecoverable error after 1 attempts: SMTP not configured, check your grafana.ini config file's [smtp] section\"" logger=alertmanager org=1
t=2022-09-08T08:11:08+0000 lvl=info msg="level=debug component=dispatcher msg=\"Received alert\" alert=testa[008ac8d][active]" logger=alertmanager org=1
t=2022-09-08T08:12:08+0000 lvl=info msg="level=debug component=dispatcher msg=\"Received alert\" alert=testa[008ac8d][active]" logger=alertmanager org=1
t=2022-09-08T08:12:10+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:12:10+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=2 size=23 referer="http://10.0.2.1:8080/alerting/grafana/PT9b_AM4k/view?returnTo=%2Falerting%2Flist"
t=2022-09-08T08:12:11+0000 lvl=eror msg="API error" logger=context userId=1 orgId=1 uname=admin error="Failed to evaluate queries and expressions: failed to execute conditions: plugin unavailable" remote_addr=10.42.0.1
t=2022-09-08T08:12:11+0000 lvl=info msg="Request Completed" logger=context userId=1 orgId=1 uname=admin method=POST path=/api/v1/eval status=400 remote_addr=10.42.0.1 time_ms=3 size=23 referer="http://10.0.2.1:8080/alerting/grafana/PT9b_AM4k/view?returnTo=%2Falerting%2Flist"
t=2022-09-08T08:13:08+0000 lvl=info msg="level=debug component=dispatcher msg=\"Received alert\" alert=testa[008ac8d][active]" logger=alertmanager org=1
t=2022-09-08T08:14:08+0000 lvl=info msg="level=debug component=dispatcher msg=\"Received alert\" alert=testa[008ac8d][active]" logger=alertmanager org=1
```

### Steps to Reproduce

1) Open Grafana
2) Create a new alert
[image]
3) You will get errors

### Context (Environment)

Affets DanielH from our community discord.

### Version

v0.9.x

Scarjit, 2022-09-08:
https://github.com/grafana/grafana/issues/46347

JeremyTheocharis, 2022-09-08:
Our current datasource does not allow alerting. This requires the go version. During the historian implementation we will enable alerting.

closed 2024-10-22 by JeremyTheocharis