united-manufacturing-hub/united-manufacturing-hub #1244: Unify logging handling

Scarjit, 2022-09-08:
**Is your feature request related to a problem? Please describe.**
Currently every services has it's own method to setup zap logging.

**Describe the solution you'd like**
Move the code to a new module (this should be inside a new git repo, so our non-core services can also use it) and fix all services to use it.

DaruZero, 2022-09-08:
Should that module be inside the umh-lib repo? Given that it should be for common code shared between our repositories, I think it should fit.
Otherwise I can create a umh-utils repo and upload the new module (and eventually other ones) in that

Scarjit, 2022-09-08:
@DaruZero The umh-lib was an experiment, but it is quite outdated, just make a new repo

DaruZero, 2022-09-08:
Is umh-utils a suitable name?

closed 2022-09-09 by Scarjit

JeremyTheocharis, 2022-09-12:
Discussion: structured, leveled logging - https://github.com/golang/go/discussions/54763

PR #1246 (2022-09-09): refactor: unify logging handling using umh-utils library
# Description

This PR unifies the logger initialization for all services

Fixes #1244 

## Type of change

Refactor: the code has changed its structure but not its behavior

# How Has This Been Tested?

The `custom-microservice-tester` image builds without errors

# Checklist:

- [ ] I have made corresponding changes to the documentation
- [x] I have made corresponding changes to the changelog (if needed).
- [ ] I have made corresponding notices to the upgrade instructions (if needed)