united-manufacturing-hub/united-manufacturing-hub #1258: Append pV tag to grafana view

Scarjit, 2022-09-19:
**Is your feature request related to a problem? Please describe.**
Currently you need to name the processValues like this "laser1distance", to be able to differentiate between different sensors on an asset.

**Describe the solution you'd like**
Our stack already partly supports using an tag after the processValue inside the topic name (`ia/factoryinsight/testLocation/testMachine/processValue/laser1distance`).
This should be extended, so that grafana shows:
`process_TAG_Name`

Example:

```
msg.payload = {
    "timestamp_ms": Date.now(), 
    "laser1distance": parseInt(msg.payload)
}
msg.topic = "ia/factoryinsight/testLocation/testMachine/processValue/abc"
return msg;
```
Should be `process_abc_laser1distance`

Scarjit, 2022-09-19:
Maybe also split it in the Grafana view to have a selector for Location, Asset, Value (but value only shows processValue)
If pV is selected show a selector for Tag and name

JeremyTheocharis, 2022-09-20:
let's combine it with the historian work stream of @DaruZero and discuss potential other implementations. let's call this feature also "grouping tag values"

The current approach is not ideal if you want to use a processValue/XXX topic without using a group

closed 2022-12-21 by Scarjit