united-manufacturing-hub/united-manufacturing-hub #1266: Using quotation marks in processValues will not return an error, but will also not save the message

JeremyTheocharis, 2022-09-20:
This message here will not be stored, but it will also not throw an error message in kafka-to-postgresql. The message was added via Node-RED into MQTT. 

[image]

See also:
https://umhhelp.zendesk.com/agent/tickets/136

Scarjit, 2022-09-21:
Can you show me the actual json ?

JeremyTheocharis, 2022-09-29:
You see the json above in the image. just think of the required {} and ,

{
  "timestamp_ms": 56465465,
....
}

Scarjit, 2022-09-29:
@JeremyTheocharis The problem is, that I don't trust the js console print to represent the generated json correctly.

But the real reason here is, that a string is not an integer (at least not for go), therefor it was dropped, due to being unparsable.

JeremyTheocharis, 2022-09-30:
Can we ensure that we get error messages in these cases where messages are dropped due to being unparsable?

PR #1299 (2022-10-11): Try convert string to float64 in kafka-to-postgresql
# Description

This pr adds functionality to kafka-to-postgres, to try parse non-float64 values as float64. (for example strings containing floats).
It also increases the log level for incorrect messages to WARN

See: 
 - https://go.dev/ref/spec#Integer_literals
 - https://go.dev/ref/spec#Floating-point_literals
 - https://forum.golangbridge.org/t/potential-bug-in-strconv-parsefloat/29079

Fixes #1266 

## Type of change

- [X] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

- [X] Test nodered flow with float [... cut]
closed 2022-10-11 by Scarjit