united-manufacturing-hub/united-manufacturing-hub #1286: Edit the timescale database, to allow our new datamodel changes

Scarjit, 2022-10-04:
**Is your feature request related to a problem? Please describe.**
The current db, can't handle all fields of the new datamodel

**Describe the solution you'd like**
Check which additional fields / tables are needed, and create them.
Create an SQL script to migrate to the new database model

Scarjit, 2022-10-05:
This is also a good time, to bring our db further to the 3rd normal form

Scarjit, 2022-10-05:
Proposed changes:
```
// ISA 95 hierachy
Table enterpriseTable {
  id int [pk]
  name string [not null, unique]
}
Table siteTable {
  id int [pk]
  name string [not null]
  enterprise int [ref: > enterpriseTable.id]
}
Table areaTable {
  id int [pk]
  name string [not null]
  site int [ref: > siteTable.id]
}
Table productionLineTable {
  id int [pk]
  name string [not null]
  area int [ref: > areaTable.id]
}
Table workCellTable {
  id int [pk]
  name string [not null]
  productionLine int [ref: > productionLineTable.id]
}

/*
  assetId will be replaced by workCellTable.id in all tables
  assetTable will be removed
*/

// New processValueTable && processValueStringTable

Table processValueNameTable {
  id int [pk]
  name string [not null, unique]
}

Table processValueTable {
  timestamp timestamp [not null]
  workCell int [ref: > workCellTable.id]
  valueName int  [not null, ref: > processValueNameTable.id]
  value double  [not null]
}

Table processValueStringTable {
  timestamp timestamp [not null]
  workCell int [ref: > workCellTable.id]
  valueName int  [not null, ref: > processValueNameTable.id]
  value string  [not null]
}

```

View formatted: https://dbdiagram.io/d/633d9c65f0018a1c5f9ed3d4

DaruZero, 2022-10-06:
Should `valueName` in tables `processValueTable` and `processValueStringTable` be `valueNameId` instead?

Scarjit, 2022-10-06:
yes

Scarjit, 2022-10-06:
same with all other refs

Scarjit, 2022-10-07:
@JeremyTheocharis I tried working on an SQL script converting the old to the new datamodel yesterday, but i think it is time, to create a microservice for version upgrades. (See #1295 )

PR #1294 (2022-10-13): Timescale changes for new datamodel
# Description

This pr includes changes to the datamodel, making it more ISA95 compliant.

Fixes #1286 

## Type of change

- [X] New feature (non-breaking change which adds functionality)
- [X] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [X] This change requires a documentation update

# How Has This Been Tested?

- [x] Installs correctly
- [ ] Migrations from old datamodel work

# Checklist:

- [ ] I have made corresponding changes to the documentation
- [ ] I have made corresponding changes to the cha [... cut]
closed 2022-10-13 by Scarjit