united-manufacturing-hub/united-manufacturing-hub #1306: Inconsistent datatypes in message structs

Sphingobium, 2022-10-12:
### Expected Behavior

timestamp_ms and timestamp_ms_end should consistently have the same data type, either uint64 or uint32

### Current Behavior

uint32 and uint64 are used interchangibly

### Steps to Reproduce

.

### Context (Environment)

for example kafka-to-postgresql/pxAddShift.go and pxDeleteShift.go use different data types for their timestamp

### Possible Solution

decide on a datatype, make sure they are consistent and maybe even write a new document or file about which variable should have which data type across the board so there are no issues. 

### Version

v0.9.x

Scarjit, 2022-10-12:
I guess we should pick uint64.
While uint32 will last until February 7, 2106, that's only 84 years.

Sphingobium, 2022-10-13:
Also not that much of an issue but still inconsistent: the key "state" is usually uint64 while the key "NewState" is a uint32 value in ModifyState.

closed 2024-10-22 by JeremyTheocharis