united-manufacturing-hub/united-manufacturing-hub #1317: helm install error

akashtalole, 2022-10-15:
### Expected Behavior

umh should run successfully

### Current Behavior

getting error

### Relevant log output

```shell
`Error: INSTALLATION FAILED: unable to build kubernetes objects from release manifest: [resource mapping not found for name: "united-manufacturing-hub-grafana" namespace: "" from "": no matches for kind "PodSecurityPolicy" in version "policy/v1beta1"
ensure CRDs are installed first, resource mapping not found for name: "united-manufacturing-hub-grafana" namespace: "united-manufacturing-hub" from "": no matches for kind "PodDisruptionBudget" in version "policy/v1beta1"
ensure CRDs are installed first, resource mapping not found for name: "united-manufacturing-hub-vernemq" namespace: "united-manufacturing-hub" from "": no matches for kind "PodDisruptionBudget" in version "policy/v1beta1"
ensure CRDs are installed first]
                                 `
```

### Steps to Reproduce

`helm install --values https://umh.app/development.yaml united-manufacturing-hub ./deployment/united-manufacturing-hub/ -n united-manufacturing-hub`

### Context (Environment)

Windows11

### Version

v0.9.x

Scarjit, 2022-10-15:
Hi,

can you tell me what your installation target is (minikube, k3os,...) ?

Regards
Ferdinand

t4talole, 2022-10-15:
its minikube on Windows11

akashtalole, 2022-10-15:
It's minikube.

akashtalole, 2022-10-15:
> Hi,
> 
> can you tell me what your installation target is (minikube, k3os,...) ?
> 
> Regards
> Ferdinand

Minikube

JeremyTheocharis, 2022-10-15:
Hi,

I already replied to you on Discord as well. I think our installation guide https://learn.umh.app/guides/umh/installation/tutorial/local-minikube-installation/ is not correct. 

Please try the VM guide instead: https://learn.umh.app/guides/umh/installation/tutorial/installation-vm/

JeremyTheocharis, 2022-10-15:
If you want minikube, you can use this tutorial in the meantime: https://united-manufacturing-hub.notion.site/Local-minikube-installation-9fec99d665794ef3aa94f0dd5bf23d49

This will work with minikube v1.26.1 on Microsoft Windows 11 Pro 10.0.22000 Build 22000

akashtalole, 2022-10-15:
I will try it.

akashtalole, 2022-10-16:
> If you want minikube, you can use this tutorial in the meantime: https://united-manufacturing-hub.notion.site/Local-minikube-installation-9fec99d665794ef3aa94f0dd5bf23d49
> 
> This will work with minikube v1.26.1 on Microsoft Windows 11 Pro 10.0.22000 Build 22000
> 
> [image]

I tried with steps mentioned in this link. I am getting same error.

Scarjit, 2022-10-16:
Ill be back home in a couple hours.
Can you tell me which minikube, helm & kubectl version you are using ?

akashtalole, 2022-10-17:
> Ill be back home in a couple hours.
> Can you tell me which minikube, helm & kubectl version you are using ?
> 

minikube version: v1.27.1
helm version: v3.10.0
kubectl version: 1.25

JeremyTheocharis, 2022-10-18:
Hi @akashtalole,

we are trying to reproduce the problem, but struggle to do so. The only way we can is by using an old Kubernetes version. Did you have minikube already installed previously? Because using choco I cannot get to your version, so it must mean you have been using it before and maybe installed it from another source. Try doing a `minikube delete` and then `minikube start ...` again, to reset the Kubernetes version.

You can also check the version using `kubectl version --short`. For me it says v1.24.3 for the server.

akashtalole, 2022-10-18:
` minikube start ...
* minikube v1.26.1 on Microsoft Windows 11 Enterprise 10.0.22000 Build 22000
* minikube 1.27.1 is available! Download it: https://github.com/kubernetes/minikube/releases/tag/v1.27.1
* To disable this notice, run: 'minikube config set WantUpdateNotification false'

* Using the docker driver based on user configuration
* Using Docker Desktop driver with root privileges
* Starting control plane node minikube in cluster minikube
* Pulling base image ...
* Downloading Kubernetes v1.24.3 preload ...`

akashtalole, 2022-10-18:
seems it now working fine
` helm install united-manufacturing-hub united-manufacturing-hub/united-manufacturing-hub -n united-manufacturing-hub
W1018 13:25:11.837828   25372 warnings.go:70] policy/v1beta1 PodSecurityPolicy is deprecated in v1.21+, unavailable in v1.25+
W1018 13:25:11.840567   25372 warnings.go:70] policy/v1beta1 PodDisruptionBudget is deprecated in v1.21+, unavailable in v1.25+; use policy/v1 PodDisruptionBudget
W1018 13:25:11.844818   25372 warnings.go:70] policy/v1beta1 PodDisruptionBudget is deprecated in v1.21+, unavailable in v1.25+; use policy/v1 PodDisruptionBudget
W1018 13:25:12.472505   25372 warnings.go:70] policy/v1beta1 PodSecurityPolicy is deprecated in v1.21+, unavailable in v1.25+
W1018 13:25:12.477859   25372 warnings.go:70] policy/v1beta1 PodDisruptionBudget is deprecated in v1.21+, unavailable in v1.25+; use policy/v1 PodDisruptionBudget
W1018 13:25:12.477859   25372 warnings.go:70] policy/v1beta1 PodDisruptionBudget is deprecated in v1.21+, unavailable in v1.25+; use policy/v1 PodDisruptionBudget
NAME: united-manufacturing-hub
LAST DEPLOYED: Tue Oct 18 13:25:07 2022
NAMESPACE: united-manufacturing-hub
STATUS: deployed
REVISION: 1
NOTES:
TODO`

JeremyTheocharis, 2022-10-18:
Do you know what you changed that it worked?

akashtalole, 2022-10-19:
Downgraded Kubernetes to v1.24.3

JeremyTheocharis, 2022-10-19:
We were able to identify the cause. If you install minikube not using choco (as recommended in the tutorial), then you might get a newer version of Kubernetes resulting in above error. 

The solution is to delete your current minikube cluster and install minikube v1.26.1, e.g., using the following command:

```
curl -LO https://storage.googleapis.com/minikube/releases/v1.26.1/minikube-linux-amd64 && sudo install minikube-linux-amd64 /usr/local/bin/minikube
```

@Sphingobium

closed 2022-10-19 by JeremyTheocharis

akashtalole, 2022-10-19:
I installed minikube 1.26.1 using this command `choco install minikube  --version=1.26.1`