united-manufacturing-hub/united-manufacturing-hub #1406: kafka-to-postgresql: Postgres not yet available: pq: database "sslmode=true" does not exist

desertspider, 2023-01-06:
### Expected Behavior

I am testing kafka-to-postgresql. I get the following Error:

kafka-to-postgresql: Postgres not yet available: pq: database "sslmode=true" does not exist 

My docker compose:

```
  kafka-to-postgresql:
    image: a0aa3649a177
    restart: always
    environment:
     - DRY_RUN=false
     - KAFKA_BOOTSTRAP_SERVER=kafka:9092
     - KAFKA_LISTEN_TOPIC=^ia.+
     - POSTGRES_HOST='azuretenant.postgres.database.azure.com'
     - POSTGRES_PORT=5432
     - POSTGRES_SSLMODE=true
     - POSTGRES_USER=username
     - POSTGRES_PASSWORD=password
     - POSTGRES_DB=FACTORYINSIGHT_DATABASE
    depends_on:
     - kafka
```
POSTGRES_SSLMODE to false gives the same error, only with database sslmode=false

Clearly it uses the ssl mode als database name.

I pulled the latest code from git and compiled all the images with "compile_all_containers.sh"

If i use the Postgres 12 default docker compose database, i get errors that SSL is not enabled (even if i set SSLMODE to false)

### Current Behavior

.

### Steps to Reproduce

.

### Context (Environment)

.

### Version

not applicable

Scarjit, 2023-01-06:
The key should be `POSTGRES_DATABASE` instead of `POSTGRES_DB`.

We should make it give out an better error if it not set.

desertspider, 2023-01-06:
Good catch. I just copied the compose commands from
deployment/kafka-to-postgresql/docker-compose.yml 
without checking the ENV variables. Changing to POSTGRES_DATABASE fixes the problem.

closed 2024-10-22 by JeremyTheocharis