united-manufacturing-hub/united-manufacturing-hub #1412: kafka-to-postgresql wont start when only using processValue

desertspider, 2023-01-09:
### Expected Behavior

When testing kafka-to-postgresql, it wont start when only using a processValue messages. Error message topic dont exist. When sending one message in for example count, the count topic is created and kafka-to-postgresql starts and then also will proces the processValue messages. 

A minor bug.

### Current Behavior

.

### Steps to Reproduce

.

### Context (Environment)

.

### Version

v0.9.x

Scarjit, 2023-01-09:
Can you give me an excerpt of the kafka-to-postgresql logs ?

desertspider, 2023-01-10:
This is the log when there are no topics in Kafka:

> This is kafka-to-postgresql build date: 2023-01-06 20:38:27 | log.level=info @timestamp=2023-01-10T08:11:49.089Z log.origin={"file.name":"kafka-to-postgresql/main.go","file.line":35} ecs.version=1.6.0
> 
> %4|1673338309.391|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.391|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property enable.auto.commit is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.391|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property enable.auto.offset.store is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.391|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property auto.offset.reset is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.405|CONFWARN|rdkafka#producer-3| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.405|CONFWARN|rdkafka#producer-3| [thrd:app]: Configuration property enable.auto.commit is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.405|CONFWARN|rdkafka#producer-3| [thrd:app]: Configuration property enable.auto.offset.store is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.405|CONFWARN|rdkafka#producer-3| [thrd:app]: Configuration property auto.offset.reset is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.406|CONFWARN|rdkafka#producer-5| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.406|CONFWARN|rdkafka#producer-5| [thrd:app]: Configuration property enable.auto.commit is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.406|CONFWARN|rdkafka#producer-5| [thrd:app]: Configuration property auto.offset.reset is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.424|CONFWARN|rdkafka#producer-6| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.424|CONFWARN|rdkafka#producer-6| [thrd:app]: Configuration property enable.auto.commit is a consumer property and will be ignored by this producer instance
> 
> %4|1673338309.424|CONFWARN|rdkafka#producer-6| [thrd:app]: Configuration property auto.offset.reset is a consumer property and will be ignored by this producer instance
> 
> Allowed memory size is 1073741824 | log.level=info @timestamp=2023-01-10T08:11:49.425Z log.origin={"file.name":"kafka-to-postgresql/main.go","file.line":180} ecs.version=1.6.0
> 
> Initializing cache with DB cache size 268435456 bytes | log.level=info @timestamp=2023-01-10T08:11:49.425Z log.origin={"file.name":"kafka-to-postgresql/cache.go","file.line":15} ecs.version=1.6.0
> 
> Performance report| Commits: 0.000000, Commits/s: 0.000000| Messages: 0.000000, Messages/s: 0.000000| PutBacks: 0.000000, PutBacks/s: 0.000000| Confirmed: 0.000000, Confirmed/s: 0.000000| [HI] Processor queue length: 0| [HI] PutBack queue length: 0| [HI] Commit queue length: 0| Messagecache hitrate 0.000000| Dbcache hitrate 0.000000| [HT] ProcessValue queue length: 0| [HT] ProcessValueString queue length: 0| [HT] Processor queue length: 0| [HT] PutBack queue length: 0 | log.level=info @timestamp=2023-01-10T08:11:49.444Z log.origin={"file.name":"kafka-to-postgresql/main.go","file.line":377} ecs.version=1.6.0
> 
> [TP] Unknown topic or partition: Subscribed topic not available: umh.v1.kafka.newTopic: Broker: Unknown topic or partition | log.level=error @timestamp=2023-01-10T08:11:49.444Z log.origin={"file.name":"internal/kafkaSharedTypes.go","file.line":156} ecs.version=1.6.0
> 
> panic: runtime error: invalid memory address or nil pointer dereference
> 
> [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x80872d]
> 
> goroutine 81 [running]:
> 
> github.com/united-manufacturing-hub/united-manufacturing-hub/internal.waitNewMessages({0xf069bf, 0x4}, 0x0?, 0x0)
> 
> 	/build/internal/kafkaSharedTypes.go:157 +0x18d
> 
> github.com/united-manufacturing-hub/united-manufacturing-hub/internal.ProcessKafkaTopicProbeQueue({0xf069bf, 0x4}, 0x0?, 0x0?)
> 
> 	/build/internal/kafkaSharedTypes.go:121 +0x7a
> 
> created by main.main
> 
> 	/build/cmd/kafka-to-postgresql/main.go:241 +0x1597

This is the log when there are only processValue topics created by mqtt-kafka-bridge:

> This is kafka-to-postgresql build date: 2023-01-06 20:38:27 | log.level=info @timestamp=2023-01-10T08:15:12.391Z log.origin={"file.name":"kafka-to-postgresql/main.go","file.line":35} ecs.version=1.6.0
> 
> %4|1673338512.781|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance
> 
> %4|1673338512.781|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property enable.auto.commit is a consumer property and will be ignored by this producer instance
> 
> %4|1673338512.781|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property enable.auto.offset.store is a consumer property and will be ignored by this producer instance
> 
[... cut ...]
> Closing database connection | log.level=info @timestamp=2023-01-10T08:15:22.174Z log.origin={"file.name":"kafka-to-postgresql/database.go","file.line":89} ecs.version=1.6.0
> 
> Successful shutdown. Exiting. | log.level=info @timestamp=2023-01-10T08:15:22.174Z log.origin={"file.name":"kafka-to-postgresql/main.go","file.line":353} ecs.version=1.6.0

Following topics exist in Kafka:

- ia
- ia.1.1.1.processValue (with one record)
- - {"timestamp_ms":1673338467807,"energyConsumption":123456}
- umh.v1.kafka.newTopic (with two records):
- - {"topic":"ia.1.1.1.processValue"}
- - {"topic":"ia"}

Scarjit, 2023-01-10:
Ok, ill have to document that umh.v1.kafka.newTopic is required for this service to start.

Scarjit, 2023-01-10:
See also: https://github.com/united-manufacturing-hub/learn.umh.app/pull/200

desertspider, 2023-01-10:
> Ok, ill have to document that umh.v1.kafka.newTopic is required for this service to start.

Even with umh.v1.kafka.newTopic it wont start, see first quote. When ia.1.1.1.Count is created it will start.

PR #1415 (2023-01-20): feat: each microservice will now attempt to create the probeTopic
# Description

This pr makes our services automatically create the probeTopic.

Fixes #1412 

## Type of change

- [X] Bug fix (non-breaking change which fixes an issue)
closed 2024-10-22 by JeremyTheocharis