united-manufacturing-hub/united-manufacturing-hub #1413: High CPU when using mqtt-to-kafka

desertspider, 2023-01-09:
### Expected Behavior

When starting mqtt-kafka-bridge, both mqtt-kafka-bridge and kafka micro services in docker will show a CPU usage of around 50%. Even if there are no MQTT messages sent. This is with no micro serves other than the mentioned two.

When using telegraf with a OPC UA demo server to kafka the CPU will stay a only a few percent. 

I guess a for loop is causing high CPU. Is this by design? 

Not a bug perse, but wasted watts/cpu cycles is still a waste :)

### Current Behavior

.

### Steps to Reproduce

.

### Context (Environment)

.

### Version

v0.9.x

Scarjit, 2023-01-09:
Are there any logs ?

desertspider, 2023-01-10:
Here is the log:

I am running the latest Kafka version.

`This is mqtt-kafka-bridge build date: 2023-01-06 20:38:55 | log.level=info @timestamp=2023-01-10T08:08:18.680Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":36} ecs.version=1.6.0

Running in Kubernetes mode%!(EXTRA string=MQTT-Kafka-Bridge, string=ia/#) | log.level=info @timestamp=2023-01-10T08:08:18.692Z log.origin={"file.name":"mqtt-kafka-bridge/mqtt.go","file.line":134} ecs.version=1.6.0

Connected to MQTT broker%!(EXTRA string=MQTT-Kafka-Bridge) | log.level=info @timestamp=2023-01-10T08:08:18.693Z log.origin={"file.name":"mqtt-kafka-bridge/mqtt.go","file.line":102} ecs.version=1.6.0

MQTT subscribed%!(EXTRA string=ia/#) | log.level=info @timestamp=2023-01-10T08:08:18.693Z log.origin={"file.name":"mqtt-kafka-bridge/mqtt.go","file.line":163} ecs.version=1.6.0

%4|1673338098.700|CONFWARN|rdkafka#producer-2| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance

%4|1673338098.704|CONFWARN|rdkafka#producer-3| [thrd:app]: Configuration property group.id is a consumer property and will be ignored by this producer instance

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:18.711Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:23.712Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:28.712Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:33.713Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:38.714Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:43.714Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:48.715Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0

Reporting stats| MQTT->Kafka queue length: 0| Kafka->MQTT queue length: 0| Produces Kafka messages: 0.000000| Produces Kafka messages/s: 0.000000 | log.level=info @timestamp=2023-01-10T08:08:53.716Z log.origin={"file.name":"mqtt-kafka-bridge/main.go","file.line":218} ecs.version=1.6.0`

desertspider, 2023-01-10:
Strange.

My docker-compose:

>   zookeeper:
>     container_name: zookeeper
>     image: 'bitnami/zookeeper:latest'
>     restart: unless-stopped
>     environment:
>       - ALLOW_ANONYMOUS_LOGIN=yes
>     volumes:
>       - "zookeeper_data:/bitnami"
>   mqtt-kafka-bridge:
>     image: deploymentmqtt-kafka-bridge:latest
>     container_name: mqtt-kafka-bridge
>     restart: unless-stopped
>     environment:
>      - MQTT_BROKER_URL=192.168.0.152:1883
>      - MQTT_TOPIC=ia/#
>      - MQTT_CERTIFICATE_NAME=NO_CERT
>      - MY_POD_NAME=MQTT-Kafka-Bridge
>      - KAFKA_BOOTSTRAP_SERVER=kafka:9092
>      - KAFKA_LISTEN_TOPIC=ia.+
>      - KAFKA_BASE_TOPIC=ia
>     depends_on:
>      - kafka
>   kafka:
>     container_name: kafka
>     image: 'bitnami/kafka:latest'
>     restart: unless-stopped
>     ports:
>       - 9092:9092
>     volumes:
>       - "kafka_data:/bitnami"
>     environment:
>       - KAFKA_CFG_ZOOKEEPER_CONNECT=zookeeper:2181
>       - ALLOW_PLAINTEXT_LISTENER=yes
>       - KAFKA_LISTENERS=INTERNAL://0.0.0.0:9092,OUTSIDE://0.0.0.0:9094
>       - KAFKA_ADVERTISED_LISTENERS=INTERNAL://kafka:9092,OUTSIDE://localhost:9094
>       - KAFKA_LISTENER_SECURITY_PROTOCOL_MAP=INTERNAL:PLAINTEXT,OUTSIDE:PLAINTEXT
>       - KAFKA_INTER_BROKER_LISTENER_NAME=INTERNAL
>     depends_on:
>       - zookeeper

docker stats

[image]

Scarjit, 2023-01-12:
There is no way to turn of bidirectional transfer, but you can "disable" it, by listening to an always empty topic.

PR #1422 (2023-01-20): feat: added fgtrace
# Description

This PR introduces [fgtrace](https://github.com/felixge/fgtrace), instead of pprof, for easier debugging.
It also changes the behavior of mqttkafkabridge, to use an exponential backoff.

Fixes #1413 

## Type of change

- [X] Bug fix (non-breaking change which fixes an issue)
- [X] New feature (non-breaking change which adds functionality)