united-manufacturing-hub/united-manufacturing-hub #1607: [FEATURE] NATS Message broker

gedw99, 2023-05-31:
Nats is like Kafka but golang . 

Has mqtt support 

works great with benthos.

lower cou , easier to run and manage. No jvm etc.
Easy to upgrade on the fly using lane duck mode.

nsc for for security and integration 

anyways if you think it’s a good idea let me know.

Scarjit, 2023-05-31:
Do you know how NATS compares to Redpanda in terms of resource usage ?
For example, if i have a broker with ~500-1000 msg/sec, what is the CPU & Memory usage of the NATS server ?

gedw99, 2023-06-09:
hey @Scarjit 

I have run Redpanda and kafka and nats jetstream. Before either existed ran tibco rendezvous :) 

https://www.youtube.com/watch?v=C4BnJ5QLeTY is a good comparison between kafka and nats that goes into the type of detail you might care about....

NATS is 
- lighter
- easier to do ops at day 2. lame duck mode :)
- more flexible.
- - much more scalable for global 
- has LEAF NODES that is designed for IOT !!!

JeremyTheocharis, 2023-06-26:
Thank you for the suggestion! At the moment, we will stick with MQTT and Kafka.

We need an MQTT in any case, as MQTT is very common in IoT and on the shopfloor. For stream processing, we leverage additionally Kafka (see also our reasoning here: https://learn.umh.app/blog/tools-techniques-for-scalable-data-processing-in-industrial-iot/)

gedw99, 2023-06-26:
all good :) wil close...

gedw99, 2023-06-26:
@Scarjit 

read that article. YOu know NATS has Leaf mode, which is for IOT factories etc where a network partition does not stop everything.

Scarjit, 2023-06-26:
@gedw99 We are currently in the process of restructuring our datamodel.
I would therefor not change the broker, until we have finalized the new datamodel.