united-manufacturing-hub/united-manufacturing-hub #1628: Management Console won't open

lmcemaj, 2023-08-15:
### Expected Behavior 🌈

The Management Console should open up, when I click the "UHM Management Console.exe.

### Current Behavior 💥

I've tried installing on three different machines, but the program won't open. When I press the symbol nothing happens.

### Steps to Reproduce 🔍

Step 1: Install "UMH Management Console_latest_x64_en-US.exe"
Step 2: Click the shortcut "UMH Management Console"

### Context (Environment) 🌎

Windows 10 machine.

### Version 📦

v0.9.x

Scarjit, 2023-08-15:
Thanks for the report.
Can you send us the logs of the management console (%localappdata%\management-console) ?

Either here, or per e-mail at fl@umh.app

lmcemaj, 2023-08-15:
> Thanks for the report. Can you send us the logs of the management console (%localappdata%\management-console) ?
> 
> Either here, or per e-mail at [fl@umh.app](mailto:fl@umh.app)

[00:00:00.061] (1674) ERROR  thread 'main' panicked at 'Failed to spawn sidecar: Io(Os { code: 2, kind: NotFound, message: "The system cannot find the file specified." })': src-tauri\src\main.rs:199
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: <unknown>
  16: BaseThreadInitThunk
  17: RtlUserThreadStart

Scarjit, 2023-08-15:
I have found & fixed the bug.
Here is a new version, which includes the bug fix:
https://download.mgmt.docs.umh.app/UMH%20Management%20Console_0.3.102_x64_en-US.exe

Please note that this version, contains some features (The Datasources tab), which are not yet released and might contain bugs.

Scarjit, 2023-08-18:
This fix is now also released offically

closed 2023-08-18 by Scarjit